adafruit/Adafruit_Python_CharLCD #9: No events detection for buttons on RPi

benit2, 2015-04-27:
Just bought this product: http://www.adafruit.com/product/1110
It works like a charm but something is bugging me..
Right now it seems that the only possible way to detect button presses is by using is_pressed() on a busy waiting loop. Needless to say this is highly inefficient.
This request involves 2 projects (this one and Adafruit_GPIO). I've looked into both the code of this project and of Adafruit_GPIO, and it seems that the use of  `GPIO.RPiGPIOAdapter` in the MCP (instead of `GPIO.BaseGPIO`) would be more appropriate in the case of using a RPi, because it supports the events features not included in the current code. 

I've tried to do this myself but I don't quite understand the code in `Adafruit_GPIO/MCP230xx.py` since it's my first time playing around the gpio. I'm not quite sure if it's a valid request or if I should submit it on the Adafruit_GPIO project instead.

ladyada, 2015-04-27:
yeah ideally we should have designed it with an interrupt output :/

tdicola, 2015-04-27:
Ah yep looking at the schematic unfortunately the interrupt pins aren't routed to GPIO on the Pi that we could read.  However, I wouldn't fret too much about the performance of polling for button presses.  Since the calls are made over the I2C bus which only runs at 100khz the Pi CPU will be waiting for IO from the bus most of the time so there shouldn't be too much of a CPU hit while polling.

closed 2015-04-27 by tdicola

benit2, 2015-04-28:
Thank you for the quick response.
I didn't know about the I2C restrictions, but still, thank you for checking it up.
Hmm, I was planning to use my Pi as a headless mini server, using this display as a limited physical interface. That's why I would have prefered a more cpu-efficient way to deal with inputs since they are really secondary in my case... 
I think I'll just make some sort of "input sleep mode" in my application where it takes some long sleep cycles between input checkups, until it's waked up by a button's long press.. then it'll switch to active polling... and go back to sleep mode after some time of inactivity. :)