adafruit/Adafruit_Python_CharLCD #16: Subsequent uses of message only update the second line

JanBe, 2016-01-21:
When using the `message`-method subsequently, only the second line of the LCD is updated.
An example script for reproducing the issue can be found here: https://gist.github.com/JanBe/8f8390484c13b7ab42a7

When executing the script, it would be expected to find the following output on the LCD:
```
New first line
New second line
```
However, the LCD reads:
```
First line
New second linef
```
As stated above, the first line is not updated. Additionally, the second line contains fragments from the sentence that was supposed to end up in the first line. The sentence was written to the end of the second line and then again overwritten by the updated second line:
`Second line` => `Second lineNew f` => `New second linef`

A workaround is to call `clear()` before subsequent uses of `message`, which however results in a small period of time between the two messages, in which the display is empty (and therefore looks like its flickering).

JanBe, 2016-01-21:
I considered if this may in some way be expected behaviour. With the current implementation, you can do things like:

cmd: `message('foo')`
```
foo 
 
```
cmd: `message('bar')`
```
foobar
 
```
cmd: `message('\nbaz')`
```
foobar
baz
```
But as soon as a second newline character gets involved, things get messy:

cmd: `message('lolo\ntro')`
```
foobar
trololo
```

If your original intention was to allow subsequent uses of `message` in order to append text to the current state of the LCD, maybe it would make sense to fix the issue I demonstrated in the last example and leave the basic functionality as it is. At the same time, a second method (e. g. `new_message`) could be introduced, which uses the implementation I suggested in #17.
If this wasn't your original intention, I think #17 still fits. It represents how I intuitively expected the `message`-method to work.

growmaster420, 2016-07-05:
would the home function work for your needs?

`    def home(self):
        """Move the cursor back to its home (first line and first column)."""
        self.write8(LCD_RETURNHOME)  # set cursor position to zero
self._delay_microseconds(3000) # this command takes a long time!`

JanBe, 2016-07-05:
Thanks for the suggestion!
The problem with `home` would however be the [delay of 3 seconds](https://github.com/adafruit/Adafruit_Python_CharLCD/blob/7e218daab39b88d91ba3b5574d14414bd3d68712/Adafruit_CharLCD/Adafruit_CharLCD.py#L167) after calling it, therefore we would only be able to set new messages at a very slow pace. When that is not a problem I would however prefer your suggestion to the `clear`-method I mentioned above, as it avoids the flickering.

I still think this issue is a bug though and the method is not working as expected in one or the other way, as I described in my previous comment.
Is there any chance that @ladyada may take a look at pull request #17, which fixes this issue? If you prefer a `message` and a `new_message`-method instead of adjusting the `message`-method I am happy to supply a pull request for that.

kattni, 2018-11-21:
Hello. We're going to be deprecating this library in favor of this one: https://github.com/adafruit/Adafruit_CircuitPython_CharLCD

This issue is not present in the new version of the library. I tested an updated version of your script, and it returns the desired results. The only changes I made to your script were to match the new API changes recently made to the new library.

That library is available on PyPI for you to `pip3 install adafruit-circuitpython-charlcd` and use on Raspberry Pi thanks to the Adafruit-Blinka library which should install automatically along with the other dependencies needed for using `charlcd`.  Please see the [documentation](https://circuitpython.readthedocs.io/projects/charlcd/en/latest/) for API and usage examples. There are Raspberry Pi-specific examples in the `examples/` folder in the repo.

closed 2018-11-21 by kattni