adafruit/Adafruit_Python_CharLCD #18: Backlight Off if pressing up or down buttons

sfabre1963, 2016-04-13:
In the example "sudo python char_lcd_plate.py".  Almost everything seems like it works ok, but during the button press example, if I press the up or down button the backlight turns off.  It correctly shows the button I pressed but without the backlight.  Then if I press any other button, then backlight then turns back on.    Any suggestions as to what may be causing this ?

DavidCWGA, 2016-06-17:
The example code is designed for RGB displays, but it sounds like you have a single-color display. It's trying to show a color your display doesn't have, so it just goes dark instead.

ladyada, 2018-11-21:
hiya we're clearing out old issues as we formally deprecate this library - yes that's because the example is changing the backlight colors and on mono backlights this will turn it off. lcd works fine!

closed 2018-11-21 by ladyada