adafruit/Adafruit_Python_CharLCD #19: IOError: [Errno 5] Input/output error

Karem-Elzftawy, 2016-05-06:
I'm using Raspberry pi B+ model and Adafruit 16x2 character LCD plus keypad for raspberry pi
I made all the steps in the guidelines when I write the final step sudo python cahr_lcd_plat.py the terminal gives me this error:

Traceback (most recent call last): File "char_lcd_plate.py", line 9, in lcd = LCD.Adafruit_charLCDPlate() File "build/bdist.linux-armv6l/egg/Adafruit_CharLCD/Adafruit_CharLCD.py", line 432, in init File "build/bdist.linux-armv6l/egg/Adafruit_GPIO/MCP230xx.py", line 150, in init File "build/bdist.linux-armv6l/egg/Adafruit_GPIO/MCP230xx.py", line 46, in init File "build/bdist.linux-armv6l/egg/Adafruit_GPIO/MCP230xx.py", line 130, in write_iodir File "build/bdist.linux-armv6l/egg/Adafruit_GPIO/I2C.py", line 95, in writelist IOError: [Errno 5] Input/output error

anyone can help me please? I2C operates on smbus[1]

Thanks

rokolczuk, 2016-05-14:
I had the same issue, check if you have an i2c device detected using i2cdetect. In my case in wasn't connected properly and after a reboot it started working.

tdicola, 2016-08-07:
Yep I/O error like this usually means an issue with finding or talking to the I2C device.  Make sure i2cdetect can find the device, and if not check the wiring, power to the device, etc.  Good luck!

closed 2016-08-07 by tdicola