adafruit/Adafruit_Python_CharLCD #21: Should be possible to run as non-root

mmiller7, 2016-05-19:
Currently there seems to be a number of checks (and maybe other assumptions?) that require any program using this library to run as "root" user.  This is extremely dangerous, especially if accepting user-input as it provides easy ways for an attacker to gain complete control over the operating system, possibly using it as an entry point to attack other devices and systems on the same network.

The library should instead allow you to make the calls as a non-root user and only generate an error message (or throw an exception which can be handled by the main application) if it encounters a problem.  It is possible to configure permissions such that a non-root user has access to GPIO control and then the library would be able to run as that limited user safely confined to only the permissions the program requires.

In this example, you can see that my GPIO interface can be controlled by any user in the "gpio" group.  This allows my various applications to run as unprivelaged users and the Linux kernel to enforce protections against potentially malicious activity (e.g. access user's personal files or change passwords) while still keeping full control of the GPIO interface to manipulate connected devices based on user-inputs.

` $ ls -l /sys/class/gpio/`
`total 0`
`-rwxrwx--- 1 root gpio 4096 May 17 20:35 export`
`lrwxrwxrwx 1 root gpio    0 May 17 20:35 gpio12 -> ../../devices/platform/soc/3f200000.gpio/gpio/gpio12`
`lrwxrwxrwx 1 root gpio    0 May 15 21:09 gpio17 -> ../../devices/platform/soc/3f200000.gpio/gpio/gpio17`
`lrwxrwxrwx 1 root gpio    0 May 15 21:09 gpio18 -> ../../devices/platform/soc/3f200000.gpio/gpio/gpio18`
`lrwxrwxrwx 1 root gpio    0 May 15 21:09 gpio27 -> ../../devices/platform/soc/3f200000.gpio/gpio/gpio27`
`lrwxrwxrwx 1 root gpio    0 May 15 21:09 gpio4 -> ../../devices/platform/soc/3f200000.gpio/gpio/gpio4`
`lrwxrwxrwx 1 root gpio    0 May 17 20:35 gpio6 -> ../../devices/platform/soc/3f200000.gpio/gpio/gpio6`
`lrwxrwxrwx 1 root gpio    0 Dec 31  1969 gpiochip0 -> ../../devices/platform/soc/3f200000.gpio/gpio/gpiochip0`
`-rwxrwx--- 1 root gpio 4096 May 17 20:35 unexport`

ladyada, 2016-05-19:
hiya, where specifically are the "number of checks (and maybe other assumptions?) that require any program using this library to run as root user."

mmiller7, 2016-05-19:
I'll try and trace thru the code tonight to locate it, I recall eventually finding a file that had some IF-checks for the user-ID being equal to zero (root).  I was getting an error message that specifically said "try running as root" in the message, it didn't appear to be a Python exception nor OS message.

I know I've been able to run other scripts and applications as normal users accessing GPIO without root (in fact as my normal user I am able to set a pin as output and use it from bash) but the python seems more difficult to trace thru (I'm used to C/C++ mostly).

ladyada, 2016-05-19:
the word 'root' does not appear anywhere in this repo, so closing! please open in the right location if you find it thx!

closed 2016-05-19 by ladyada

DarrellStuff, 2016-05-21:
Well, I found at least on case of checking for UID != 0 (is the running user ID the root account?) in line 49 of the FT232H.py code (Adafruit_Python_GPIO/Adafruit_GPIO/FT232H.py):

46:   def _check_running_as_root():
47:       # NOTE: Checking for root with user ID 0 isn't very portable, perhaps
48:       # there's a better alternative?
49:       if os.geteuid() != 0:
50:           raise RuntimeError('Expected to be run by root user! Try running with sudo.')

Full source at:
[https://github.com/adafruit/Adafruit_Python_GPIO/blob/cdf7a7b8a2bc49d6d37ac676a240492bdd14684b/Adafruit_GPIO/FT232H.py]

Question: Is the loading of the FT232H driver/configuration something that can be permanently installed as part of the OS when it starts and then called as needed by a non-root user account? Or is there a technical reason it always needs to be loaded by the specific Python program running under the current user ID privileges every time it is run?

andrewwade, 2018-06-12:
This still is very much an issue. Please reopen.