adafruit/Adafruit_Python_CharLCD #25: Is 8.4 Jessie supported?

MrBillium, 2016-07-04:
BB Green running Debian 8.4 Jessie
Python 3.4

I have a simple test program running that displays text on an RGB screen 
it works fine on the same hardware running Debian 7.8 Wheezy

But after upgrading to Jessie, I cannot get any text on the screen and there are no errors in non PWM mode.
In PWM mode I do get an error in Adafruit GPIO "must start the PWM channel first"

Is this library supposed to work in Jessie which has some changes in Capemgr etc?

Bill

kneerunjun, 2018-05-23:
Similar observations on my RPi 3B , running Raspbian 9.4  Jessie - double checked the GPIO connections. All what I get is 16 black boxes on the top row of my 16X2 LCD HD44780. No text ! I tried changing the data channels to other OUT GPIOs , but the same disappointing result. I did read somewhere that LCD may not be broken , but it fails to initialize HD44780 from code so tried changing my local repository head to a commit dated 2016. Still no luck - only a row of blank boxes that change contrast when I spin the trimpot on pin 3. Do I change the Distro version ?

kattni, 2018-11-21:
We're going to be deprecating this library in favor of this one: https://github.com/adafruit/Adafruit_CircuitPython_CharLCD

We're only going to be supporting this through the Blinka library. On BeagleBone, the GPIO is working with many of the new libraries using Blinka. We are currently working on more support for the BeagleBone such as I2C/SPI. Please keep an eye out for that update which will add more support for BeagleBone.

That library is available on PyPI for you to `pip3 install adafruit-circuitpython-charlcd` and use on Raspberry Pi thanks to the Adafruit-Blinka library which should install automatically along with the other dependencies needed for using `charlcd`.  Please see the [documentation](https://circuitpython.readthedocs.io/projects/charlcd/en/latest/) for API and usage examples. There are Raspberry Pi-specific examples in the `examples/` folder in the repo.

closed 2018-11-21 by kattni