adafruit/Adafruit_Python_CharLCD #28: Delay seems to not be enough time for the clear operation

corvuscrypto, 2016-12-10:
It seems that the delay for the clear functionality across CharLCDs is a tad bit too quick for at least one such device. Clearing the screen and immediately sending a message gives garbage on the screen (using an adafruit bought 16x2 character LCD display). This was resolved by increasing the delay time between the clear() and message() calls.

Please also reference [this stack overflow thread](http://stackoverflow.com/questions/41070686/weirdest-thing-in-python-if-i-dont-use-raw-input-lcd-display-gets-text-all-s#41070686) and notice the comments. This seems like a delay-time issue even after user switched to this repo's version of the lib.

- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  Raspberry Pi with Raspbian operating system

- Python version (run `python -version` or `python3 -version`):  python 2.7

- Error message you are receiving, including any Python exception traces:  N/A

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): **LIST REPRO STEPS BELOW**

1. Initialize Adafruit_CharLCD
2. Perform display clear
3. Immediately after send a message using the message class function (e.g. `lcd.message("string")`)

dusterherz, 2017-03-24:
I had exactly the same problem. I modify the sleeping and it's worked.

corvuscrypto, 2017-03-24:
I'm glad you replied because I had forgotten to follow up on this and make a PR. 

My approach will be to just allow this delay time to be overriden but set the default a bit higher than it is. For most people getting into hobby embedded projects I think an increased delay isn't a big deal.

kattni, 2018-11-21:
Hi! This issue is not present in a newer version of this library. The newer version does not inclue a settable delay, however, you can use `lcd.clear()` immediately followed by `lcd.message` without issue.

We're going to be deprecating this library in favor of this one: https://github.com/adafruit/Adafruit_CircuitPython_CharLCD

That library is available on PyPI for you to `pip3 install adafruit-circuitpython-charlcd` and use on Raspberry Pi thanks to the Adafruit-Blinka library which should install automatically along with the other dependencies needed for using `charlcd`.  Please see the [documentation](https://circuitpython.readthedocs.io/projects/charlcd/en/latest/) for API and usage examples. There are Raspberry Pi-specific examples in the `examples/` folder in the repo.

closed 2018-11-21 by kattni