adafruit/Adafruit_Python_CharLCD #29: Turning off backlight/display not working for another lcd

Arthurin, 2017-02-01:
I really like the library you made here but these 2 functions aren't working and I'm wondering if there is a way to fix this. My LCD screen is 16x2 but with only 16 pin (LED backlight).
I followed the tutorial on https://learn.adafruit.com/character-lcd-with-raspberry-pi-or-beaglebone-black/wiring except for pin 16 connected to the 5V.

- I'm using RPI 2 B with Jessie
- Python 2.7.9
- No message/traces, I'm using char_lcd.py and nothing happen when I called lcd.set_backlight(0). Same result if I replace it with lcd.enable_display(0)

I hope you can help me!
Art'

ladyada, 2017-02-01:
if you dont connect the backlight to a pin, then no you cannot turn it off

closed 2017-02-01 by ladyada

Arthurin, 2017-02-01:
the LCD pin 15 and 16 are made to power the backlight, I can't connect it with a RPI pin, can I?
I'm in the same situation than the previous tutorial : https://learn.adafruit.com/drive-a-16x2-lcd-directly-with-a-raspberry-pi/overview They said it is better to use the new char LCD python library so here I am.

ladyada, 2017-02-01:
tech support is done in the forums - not here. thx!