adafruit/Adafruit_Python_CharLCD #36: No Module Named Adafruit_CharLCD

vickey97, 2018-08-10:
Hey, I was trying to play around with LCD and installed this library. But when i write the command below in thonny and run it, it gives me this error;
ImportError: No module named 'Adafruit_CharLCD'
`
import`` time
import Adafruit_CharLCD as LCD

lcd_rs = 37
lcd_en = 35
lcd_d4 = 33
lcd_d5 = 31
lcd_d6 = 29
lcd_d7 = 23
lcd_backlight = 4

lcd_columns = 16
lcd_rows = 2

lcd = LCD.Adafruit_CharLCD(lcd_rs, lcd_en, lcd_d4, lcd_d5, lcd_d6, lcd_d7, lcd_columns, lcd_rows, lcd_backlight)

lcd.message('Hello\nworld!')
time.sleep(5.0)
lcd.clear()
text = raw_input("Type Something to be displayed: ")
lcd.message(text)

time.sleep(5.0)
lcd.clear()
lcd.message('Goodbye\nWorld!')

time.sleep(5.0)
lcd.clear()
`
Any help is greatly appreciated! Thank You

WillTWallace, 2018-08-11:
You need to ensure that all the necessary files are being accessed (be sure they are in same directory or linked properly). That error means it cannot find the file specified. Can you provide replication details?

vickey97, 2018-08-11:
Thanks for the reply WillTWallace, but I installed exactly the way "README" file said like downloading, unzipping and setup.py but this doesn't solve my problem and gives the same error "ImportError: No module named 'Adafruit_CharLCD" when i run it with Thonny Python IDE.

PS: But when I run the program through terminal window and navigate through Adafruit_Python_CharLCD/examples and run the program with sudo python LCD_Sample.py.
It seems to run the program with no error. Can you please help me out here Please?

vickey97, 2018-08-11:
Yeah! that's what caused the problem I installed Adafruit with Python2 and was trying to run the code in Python3. Now I installed this library with Python3 and it worked. Well thank you soo much!

closed 2018-08-11 by vickey97